indilib/indi-3rdparty #1127: indi ccd driver for DSLR (indi_canon_ccd) reports 0 CCD_FRAME size

artyom-beilis, 2025-08-05:
**Describe the bug**

indi_canon_ccd does not report frame size 

```
03:22:59.031   CN Canon DSLR EOS 6D:CCD_FRAME (Idle) rw type INDI_NUMBER
   0 X/Left  val=0.000000 min=0.000000 max=0.000000 step=0.000000
   1 Y/Top val=0.000000 min=0.000000 max=0.000000 step=0.000000
   2 WIDTH/Width val=0.000000 min=0.000000 max=0.000000 step=0.000000
   3 HEIGHT/Height val=0.000000 min=0.000000 max=0.000000 step=0.000000
```

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
Here an example of values reported for another Canon device

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

artyom-beilis, 2025-08-05:
Related to https://github.com/artyom-beilis/OpenLiveStacker/issues/131

knro, 2025-08-05:
That's how it normally operates. It requires the user to input the information for the first time and then it can be used for subsequent captures.

artyom-beilis, 2025-08-05:
> That's how it normally operates. It requires the user to input the information for the first time and then it can be used for subsequent captures.

How user library is expecting to know the frame size in advance? Actually if I look on log from Canon 750D from same driver there is frame size, additionally all other camera drivers report `CCD_FRAME` size

artyom-beilis, 2025-08-05:
> I usually use [Digital Camera Database](https://www.digicamdb.com/) to look up the maximum resolution and pixel size.

In such a case either this kind of DB need to be intergated to indi library or another alternative should be used (like I use in OLS) - capture a single frame and derive these values from actual frame (I use it with libgphoto2)

The problem is that _every client_ that need these values would need to implement this kind of logic

knro, 2025-08-05:
I think that's hard to do and it needs to be kept updated. Since this is usually a 1-time operation, users can usually find such information and save it.