indilib/indi-3rdparty #1139: Indi toupcap/toupbase not saving the tail light settings to config file

mireianievas, 2025-08-14:
Indi git main branch synced around 1 week ago, but also tested with the previous stable version of indi.

The tail light settings, once set in the indi device panels and saved to the config file, are not written to disk, therefore on next load they will turn on again. However, if you write them manually in the config file under $HOME/.indi/... They work and you can effectively have them off.

knro, 2025-08-14:
Tail light setting refers to what exactly?

gy-lehel, 2025-08-15:
There is a setting in the control panel of the camera, it should control the on/off status of the status LED-s in the camera backpanel. Also the control of the gain mode setting is always re-set to HCG on driver start.

gy-lehel, 2025-08-22:
I think I have found a way to save both settings, but still need to figure out, how to restore. I'll only have access to my rig next week, so it will be only then, that I'll be able to confirm that it actually works. It would also be nice if someone would be able to test it with a different camera, than the 585, just to make sure the change does not break functionality for others.

knro, 2025-08-22:
All the settings are already saved in https://github.com/indilib/indi-3rdparty/pull/1146

If you have time to compile and check that PR, it would be great.

gy-lehel, 2025-08-22:
I will some time on Monday.

gy-lehel, 2025-08-28:
Unfortunately it does not yet work fully, please follow this:
1. power on camera
2. connect indi, set tail light to off (check light, should be off
3. set it on, check light if it is on
4. set it off
5. power off camera, wait for a few seconds, power on (tail light should be the default on, and red led should be on)
6. start indi
7. setting is confirmed as saved, and restored in indi, as expected shows the tail light as off, while camera tail light is actually on.

Tested on arch linux x86_64, latest git master

baro666, 2025-08-28:
> Unfortunately it does not yet work fully, please follow this:
>
> 1. power on camera
> 2. connect indi, set tail light to off (check light, should be off
> 3. set it on, check light if it is on
> 4. set it [...]

Ach! I was ecstatic to finally be able to save the capture mode to HDR, which was what mattered most to me, and I didn't look too closely into the tail-gate. I'll run more tests.

gy-lehel, 2025-08-29:
We should also check the HDR restore, too, it is "easy" as with gain set to the 1000+ range, the noise distribution of a bias or dark frame should be quite telling. I'd expect it to be restored in the control panel, but not applied to the camera either, as when I tried to fix this issue I saw no code to restore it outside the control panel control elements.

closed 2025-09-04 by knro