indilib/indi-3rdparty #1162: Tried building indi-libcamera this weekend - only builds if I am in super user.  Maybe a permission issue?

sonny486, 2025-09-02:
Downloaded both INDI and 3rdparty drivers.

Built INDI then built 3rd party libraries.

Then tried to build indi-libcamera using:

mkdir -p ~/Projects/build/indi-libcamera
cd ~/Projects/build/indi-libcamera
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-libcamera
make -j4
sudo make install

But project would fail.

went into super user and did the same, and it built it just fine.

Weird.

sonny486, 2025-09-02:
OS Debian 12 (Orange Pi 5).

knro, 2025-09-03:
Maybe you had an existing build folder and it had some files owned by root? If you clear the build folder and try again maybe it will work.

knro, 2025-11-18:
Cannot reproduce this issue.

closed 2025-11-18 by knro